osrf/vrx #719: Update VRX Docker MWE tutorial for 2023

M1chaelM, 2023-08-15:
The tutorial for minimal working examples of competitor Docker images needs to be updated to include the 2023 image: https://github.com/osrf/vrx/wiki/tutorials-vrx_docker_mwes

M1chaelM, 2023-08-17:
Following our offline discussion yesterday, my preferred way to update this tutorial is as follows:
* Update the references to `vrx_2022_starter` image to say `vrx_2023_starter`.
* Rename the folder in `vrx-docker` that contains the 2023 image @j-herman created so it is called `vrx_2023_starter` instead of `vrx_2023_mwes`
* Delete all references to `vrx_2022_simple` 

This will mean our starter image example is slightly fancier than the image a team would produce by just following the tutorial, but I think that's Ok. Another option suggested by @caguero would be to create a `vrx_2023_simple` folder in `vrx-docker` and add the files needed.

@caguero are you ok with the first option?

caguero, 2023-08-17:
Thanks for the context, I'm totally OK with the first suggested option.

M1chaelM, 2023-08-18:
@caguero I went through to make these changes and realized I had forgotten why we had 2 images in 2022. The `simple` image is already a bit more advanced than what the teams would produce in a tutorial. We added the "starter" image because many teams were having trouble understanding how to compile a ROS workspace on their Docker image, and some of them were trying to do it at runtime.

I'm not sure whether all teams will want to build VRX on the competitor image, so maybe we need two examples after all...

caguero, 2023-09-01:
Updated in https://github.com/osrf/vrx-docker/pull/72

closed 2023-09-01 by caguero