osrf/vrx #721: Backup solution for running VRX when fuel is down

M1chaelM, 2023-08-15:
## Problem
Gazebo fuel is down a lot of the time. This is not usually a problem for users who have already run VRX and cached the issue, but it will cause VRX to fail for users who are running for the first time.

## Proposal
We need to provide another way for users to get the models they need to run VRX when fuel is down.

caguero, 2023-08-22:
I confirmed that Fuel is already running on S3 and we shouldn't experience availability issues.

I also backed up and wiped out my Fuel directory and relaunched VRX but I wasn't able to reproduce any issues. After that, I removed my my Fuel directory again and restored the original one and things worked without any redownload. I also confirmed that the sim doesn't require Internet connection when the models are already in your local cache.

M1chaelM, 2023-08-24:
Given the resolution of #726, I wonder if the problem might have been related to switching the name from `fuel.gazebosim.org` to `fuel.ignitionrobotics.org`, since systems that went through this switch would have both directories stored under `fuel`. Although it would be interesting to know, I think we can close this now that Fuel has switched to S3 and no one can currently reproduce the problem.

closed 2023-08-24 by M1chaelM