osrf/vrx #725: Gazebo crashing while training RL agent

toorxx, 2023-08-22:
I have been working with the WAM-V using VRX (Gazebo) along with OpenAI Gym to train a series of actions, and I came across the problem described in this PR #706. I implemented the proposed changes in my setup, but unfortunately, I am still facing the issue.

I am using the same version of Gazebo mentioned in this PR and have tested the WAM-V in the sydney_regatta world. Despite the WAM-V holding up for more training steps, I still experience the same error.

I wonder if you could consider reopening this case or provide me with additional guidance on how I might resolve this issue.
I am willing to collaborate on any further testing or in solving this problem if necessary. I greatly appreciate your help and your work on this project.

Thanks!

M1chaelM, 2023-08-24:
Yes, we can reopen the issue if it hasn't been fully resolved. Could you please provide the steps you're taking to reproduce the bug?

caguero, 2023-09-05:
Closing this but feel free to reopen with instructions to reproduce the bug.

closed 2023-09-05 by caguero