osrf/vrx #730: Ability to attach a manipulator

IamGyal, 2023-08-29:
Hi,

Is it possible to attach a serial manipulator underneath a USV in this simulation environment in such a way that the hydrodynamics is also applied to the manipulator?

Thank you very much.

caguero, 2023-08-29:
Hi, impossible is nothing :)

But the hydrodynamics plugin won´t automatically change the behavior of the overall USV with the presence of the manipulator. Instead, you'd have to update the parameters of the hydrodynamics to account for that.

IamGyal, 2023-08-30:
Hi, thanks for your reply :)

But could you give me a bit more explanation, please?

Do you mean the USV will behave as if there is no manipulator attached to it even when a manipulator is actually attached?

I'm looking for a situation where
1. the attached manipulator is affected by relevant hydrodynamics, 
2. the forces and torques (wrenches) propagated from the manipulator to the center of mass of the USV are taken into account, 
3. and the USV behaves correspondingly (with both the hydrodynamics for the USV and the wrenches from the manipulator being taken into account). 

Thank you very much.

IamGyal, 2023-08-30:
Or do you mean all the three points I mentioned above would happen, but just the parameters of the hydrodynamics for the resulting robot consisting of the initial USV and the manipulator should be adjusted?

caguero, 2023-08-31:
> Do you mean the USV will behave as if there is no manipulator attached to it even when a manipulator is actually attached?
>
> I'm looking for a situation where
>
> 1. the attached manipulator is affected [...]

  1. The attached manipulator won't be affected by hydrodynamics directly. It'll be indirectly because the hydrodynamics plugin attached to the USV will generate forces/torques on the `base_link` of the USV. However, the manipulator itself won't receive any forces/torques directly. Also, the force/torque applied at the USV are a direct function of the hydrodynamics parameters. In other words, the plugin will generate almost the same force/torque with or without the manipulator. 
  2. This is true and will happen thanks to the physics engine.
  3. I think this is true as well.

Maybe it's easier to identify an example where hydrodynamics won't affect the manipulator:
Imagine that your USV is still and you move your manipulator: no hydrodynamics will affect its motion, although it's moving underwater.

IamGyal, 2023-09-05:
Thank you very much for your reply.

I have further questions.

4. Would there be a way that the manipulator is also affected by hydrodynamics directly? Perhaps by applying the hydrodynamics plugin to the manipulator as well as the base?

5. Is the hydrodynamics plugin used in this simulator only for USVs? Or for any kind of rigid bodies/shapes as well?

IamGyal, 2023-09-05:
> 

Thank you very much for your reply.

I have further questions.

4. Would there be a way that the manipulator is also affected by hydrodynamics directly? Perhaps by applying the hydrodynamics plugin to the manipulator as well as the base?

5. Is the hydrodynamics plugin used in this simulator only for USVs? Or for any kind of rigid bodies/shapes as well?

IamGyal, 2023-09-05:
> 1. The attached manipulator won't be affected by hydrodynamics directly. It'll be indirectly because the hydrodynamics plugin attached to the USV will generate forces/torques on the `base_link` of [...]

Thank you very much for your reply.

I have further questions.

4. Would there be a way that the manipulator is also affected by hydrodynamics directly? Perhaps by applying the hydrodynamics plugin to the manipulator as well as the base?

5. Is the hydrodynamics plugin used in this simulator only for USVs? Or for any kind of rigid bodies/shapes as well?

caguero, 2025-05-08:
The hydrodynamics plugin can be attached to any other rigid body but we never test the behavior in a manipulator.

closed 2025-05-08 by caguero