osrf/vrx #731: How to change the boat velocity

AlexWUrobot, 2023-08-29:
**Describe the bug**
I tried to increase the boat velocity in the version 1.6.2 
However, the max velocity is still 2.4 meter / second

**Expected behavior**
What you expected to happen the velocity be higher than 2.4 meter /second

**To Reproduce**

roslaunch vrx_gazebo vrx_auto.launch 

I tried three below publish topic and all can active the boat to move but the max velocity is 2.39 meter /second, by observing the model_state_link "wamv" linear velocity 

(1) rostopic pub -r 100 /wamv/thrusters/left_thrust_cmd std_msgs/Float32 "data: 400.0"
(2) rostopic pub -r 100 /wamv/thrusters/right_thrust_cmd std_msgs/Float32 "data: 400.0"
(3) rostopic pub -r 100 /cmd_vel geometry_msgs/Twist "linear:
  x: 100.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0"

**System Configuration:**
Ubuntu 20.0, gazebo simulation, ROS 1 

**Screenshots**
I also tried to edit this three xacro file to increase the thrust from 250 N to 25000000 N , and then
**catkin build wamv_gazebo usv_gazebo_plugins**
Then, the catkin build succeeds 
Then, source the workspace
Then, roslaunch and rostopic pub. **The boat max speed just slightly increase from 2.29 m/s to 2.4 m/s**

(1) edit the parameter <maxForceFwd>250.0</maxForceFwd> in the file **wamv_gazebo_thruster_config.xacro**
[image]

(2) edit the parameter SdfParamDouble(thrusterSDF, "maxForceFwd", 250.0)  in the file **usv_gazebo_thrust_plugin.cc**
[image]

Thank you very much for any suggestion ^^

AlexWUrobot, 2023-08-30:
I find the solution, change the linear drag force in the **wamv_gazebo_dynamic_plugin.xacro** file
[image]

closed 2023-08-30 by AlexWUrobot

MustafaBayrak1, 2024-03-25:
So I tried a bunch of things such as modifying the usv_gazebo_thrust_plugin.cc, max values as you edited in wamv_gazebo_thruster_config.xacro, but nothing else worked instead of what you did at the end.

However, changing the linear drag force makes the wamv like it's going on ice. It doesn't ever stop on its own like it does normally. I hope there is a way to increase the max speed of the WAMV by using the maxForceFwd parameter.