osrf/vrx #732: Wamv model not found

huiyulhy, 2023-08-31:
**Describe the bug**
Please say what is broken. Include whether you think this is a problem with the code or the documentation / tutorials.
Running the standard competition command or command from custom WAMV to launch the simulation gives the following error on gazebo: `ros2 launch vrx_gz competition.launch.py world:=sydney_regatta`

Gazebo does not seem to be able to download the model of the WAMV - is there another way to download the meshes and place them into a directory?

**Expected behavior**
WAMV mesh should be downloaded and rendered, but the download does not seem to work. 

**To Reproduce**
Ran the same steps per the instructions on installing VRX and launching the competition environment.

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 22.04
- ROS Version: Ros2 Humble
- Gazebo Version: Gazebo garden
- Graphics Card: Nvidia 1660 super
- Are you using VRX or VRX Classic? VRX

**Screenshots**
If applicable, add screenshots to help explain your problem. (If the problem is terminal output, it is better to cut and paste the text than to take a screenshot.)

**Additional notes**
Add any other context about the problem here.

caguero, 2023-08-31:
Hi! I can't reproduce this error. WAM-V doesn't have to be downloaded, it's generated on the fly from URDF.

Could you paste the error message that you're getting?
Are you using the latest VRX version?

huiyulhy, 2023-09-01:
Sorry I forgot to attach this earlier, here is a screenshot of the error - it seems like my computer is unable to fetch the meshes from the wamv folder and also the thruster folder. I had just recently (maybe 3 days again) cloned the main branch from VRX - is that the correct VRX version? 
[image]

caguero, 2023-09-01:
Could you share the output of this command?

```
env | grep GZ
```

huiyulhy, 2023-09-02:
This is the output from my terminal for this command, thanks for helping me to troubleshoot this!
[image]

caguero, 2023-09-04:
The paths on `GZ_SIM_RESOURCE_PATH` look wrong. For instance, instead of:
`/home/lhy/vrx_ws/install/wamv_description/share/wamv_description/models`
you should have:
`/home/lhy/vrx_ws/install/share/wamv_description/models`

I wonder if you ran `colcon build` in the `vrx_ws` directory. My suggestion is to clear your workspace:
```
rm -rf build install log
```
Recompile again and in a new terminal source the appropriate `install/setup.bash`.

jjeerryy, 2023-09-04:
Hi there,

I'm getting a different issue. I'm using WSL Ubuntu 22.04.
When I run the line ros2 launch vrx_gz competition.launch.py world:=sydney_regatta
I see the gazebo sim pop up like below and then nothing shows up and it closes itself. 
[image]

I noticed there were a lot of 'server directory does not exist' when I first run the code.

Any idea why this is happening?

huiyulhy, 2023-09-05:
This works great for me now. Thanks for helping me with troubleshooting!

closed 2023-09-05 by huiyulhy

mvanlobensels, 2023-10-20:
How can I make Gazebo search in `install/wamv_description/share/wamv_description/models` instead?