osrf/vrx #736: Changing wave shader GLSL version from 330 to 150 (Ogre rendering support)

Dat-Bois, 2023-09-12:
**Describe the bug**
Ogre2 is giving significant issues when rendering, probably due to my VM and OpenGL GPU passthrough support with Gazebo. Running with Ogre as my rendering engine for other sample Gazebo environments seems to fix the rendering issues and use Gazebo as expected with GPU acceleration. However, since the shaders in the VRX project for the GerstnerWaves are written for GLSL 330 and Ogre supports only up to GLSL 150, Gazebo crashes.

**Expected behavior**
Gazebo GUI loads/opens.

**To Reproduce**
List the steps to reproduce the problem:
1. Run " ros2 launch vrx_gz competition.launch.py world:=sydney_regatta extra_gz_args:='--render-engine ogre' "
2. See error

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 22.04 ARM64 (VMware Fusion)
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Garden Sim 7.0.0
- Graphics Card: Integrated (Apple M2 Pro 19 GPU core)
- Are you using VRX or VRX Classic? VRX

**Error Log**
```
[ruby $(which gz) sim-1] gz sim server: ./OgreMain/src/OgreGpuProgramParams.cpp:1105: void Ogre::GpuProgramParameters::_writeRawConstants(size_t, const int*, size_t): Assertion `physicalIndex + count <= mIntConstants.size()' failed.
[ruby $(which gz) sim-1] Stack trace (most recent call last) in thread 37472:
[ruby $(which gz) sim-1] #25   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ruby $(which gz) sim-1] #24   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa9835d1b, in 
[ruby $(which gz) sim-1] #23   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa97cd5c7, in 
[ruby $(which gz) sim-1] #22   Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0xffffa57b31fb, in 
[ruby $(which gz) sim-1] #21   Object "/usr/lib/aarch64-linux-gnu/gz-sim-7/plugins/libgz-sim-sensors-system.so", at 0xffff7d2bda77, in gz::sim::v7::systems::SensorsPrivate::RenderThread()
[ruby $(which gz) sim-1] #20   Object "/usr/lib/aarch64-linux-gnu/gz-sim-7/plugins/libgz-sim-sensors-system.so", at 0xffff7d2bccf3, in gz::sim::v7::systems::SensorsPrivate::RunOnce()
[ruby $(which gz) sim-1] #19   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fefed2b, in gz::rendering::v7::OgreScene::PreRender()
[ruby $(which gz) sim-1] #18   Object "/lib/aarch64-linux-gnu/libgz-rendering7.so.7", at 0xffff7cfa80ef, in gz::rendering::v7::BaseScene::PreRender()
[ruby $(which gz) sim-1] #17   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe67727, in 
[ruby $(which gz) sim-1] #16   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe7056f, in 
[ruby $(which gz) sim-1] #15   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe67727, in 
[ruby $(which gz) sim-1] #14   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe7056f, in 
[ruby $(which gz) sim-1] #13   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe67727, in 
[ruby $(which gz) sim-1] #12   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe7056f, in 
[ruby $(which gz) sim-1] #11   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fe6d5d3, in 
[ruby $(which gz) sim-1] #10   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fec4077, in 
[ruby $(which gz) sim-1] #9    Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5fec8d0b, in 
[ruby $(which gz) sim-1] #8    Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5febcb7b, in gz::rendering::v7::OgreMaterial::UpdateShaderParams()
[ruby $(which gz) sim-1] #7    Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff5febc86f, in gz::rendering::v7::OgreMaterial::UpdateShaderParams(std::shared_ptr<gz::rendering::v7::ShaderParams const>, Ogre::SharedPtr<Ogre::GpuProgramParameters>)
[ruby $(which gz) sim-1] #6    Object "/lib/aarch64-linux-gnu/libOgreMain.so.1.9.0", at 0xffff5fa24127, in Ogre::GpuProgramParameters::_writeRawConstant(unsigned long, int)
[ruby $(which gz) sim-1] #5    Object "/lib/aarch64-linux-gnu/libOgreMain.so.1.9.0", at 0xffff5fa24107, in Ogre::GpuProgramParameters::_writeRawConstants(unsigned long, int const*, unsigned long)
[ruby $(which gz) sim-1] #4    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa978403f, in __assert_fail
[ruby $(which gz) sim-1] #3    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa9783fcf, in 
[ruby $(which gz) sim-1] #2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa977712f, in abort
[ruby $(which gz) sim-1] #1    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa978a67b, in raise
[ruby $(which gz) sim-1] #0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa97cf200, in 
[ruby $(which gz) sim-1] Aborted (Signal sent by tkill() 37432 1000)
[ruby $(which gz) sim-1] gz sim gui: ./OgreMain/src/OgreGpuProgramParams.cpp:1105: void Ogre::GpuProgramParameters::_writeRawConstants(size_t, const int*, size_t): Assertion `physicalIndex + count <= mIntConstants.size()' failed.
[ruby $(which gz) sim-1] Stack trace (most recent call last) in thread 37588:
[ruby $(which gz) sim-1] #31   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa25cb523, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
[ruby $(which gz) sim-1] #30   Object "/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffffa13e2f13, in g_main_context_iteration
[ruby $(which gz) sim-1] #29   Object "/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffffa143aeab, in 
[ruby $(which gz) sim-1] #28   Object "/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffffa13e5a6f, in g_main_context_dispatch
[ruby $(which gz) sim-1] #27   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa25cc057, in 
[ruby $(which gz) sim-1] #26   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa256ff87, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
[ruby $(which gz) sim-1] #25   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa256cb8f, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
[ruby $(which gz) sim-1] #24   Object "/lib/aarch64-linux-gnu/libQt5Widgets.so.5", at 0xffffa3a12abf, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[ruby $(which gz) sim-1] #23   Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa259cfd3, in QObject::event(QEvent*)
[ruby $(which gz) sim-1] #22   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff68f64923, in gz::gui::plugins::RenderThread::RenderNext(gz::gui::plugins::RenderSync*)
[ruby $(which gz) sim-1] #21   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff68f73f63, in gz::gui::plugins::RenderThreadRhiOpenGL::RenderNext(gz::gui::plugins::RenderSync*)
[ruby $(which gz) sim-1] #20   Object "/usr/lib/aarch64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0xffff68f69193, in gz::gui::plugins::GzRenderer::Render(gz::gui::plugins::RenderSync*)
[ruby $(which gz) sim-1] #19   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4e2e2b, in 
[ruby $(which gz) sim-1] #18   Object "/lib/aarch64-linux-gnu/libgz-rendering7.so.7", at 0xffff685d80ef, in gz::rendering::v7::BaseScene::PreRender()
[ruby $(which gz) sim-1] #17   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4c7727, in 
[ruby $(which gz) sim-1] #16   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4d056f, in 
[ruby $(which gz) sim-1] #15   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4c7727, in 
[ruby $(which gz) sim-1] #14   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4d056f, in 
[ruby $(which gz) sim-1] #13   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4c7727, in 
[ruby $(which gz) sim-1] #12   Object "/usr/lib/aarch64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0xffff2b4d056f, in 
[... cut ...]
I understand that I am creating an issue for an unsupported configuration (ARM64), but adding support can be incredibly useful given how many students may be using Apple M-series Macs. I am concurrently developing a guide on how to do this and will publish it once I iron out the kinks (and figure this out). 

In the meantime, if a translation can be written translating the GerstnerWaves_fs_330.glsl and GerstnerWaves_vs_330.glsl along with the relevant cpp code for glsl version 150 would be great.

matthiaskiller, 2023-10-02:
@Dat-Bois I have the same issue. Have you figured it already out? :)
Thanks!

caguero, 2025-07-07:
I cannot reproduce this issue. Perhaps it's VM-related. The command works for me and starts VRX :

```
ros2 launch vrx_gz competition.launch.py world:=sydney_regatta extra_gz_args:='--render-engine ogre'
```

[image]