osrf/vrx #739: Problem with reading the position of wam-v

KonradUr, 2023-09-26:
Hi,

for testing with my own controllers, I need information about the wamv's position.

Using ROS1 + Gazebo11 it works in general properly:

`rostopic echo /gazebo/model_states/pose[17]`

It was more difficult with ROS2 and Gazebo Garden to get it moving, but after a few more installations in the docker, thrusters also started up.

Now, in ROS2 and Gazebo Garden I don't know how to read the current position. The elements of the group 

`/wamv/pose`

return either fixed values or other than _"pose"_ in the _"component inspector"_ in GazeboSim. How to read the position of the device correctly? BTW: GPS and IMU readings are working properly.

kulkarni-raunak, 2023-11-09:
You need to enable the ground truth flag in the file [wamv_gazebo.urdf.xacro](https://github.com/osrf/vrx/blob/main/vrx_urdf/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro#L11C3-L11C3). 

```
<xacro:arg name="ground_truth_enabled" default="true" />
```

KonradUr, 2023-11-09:
Thanks for your help.
Will this modification add fields for `/wamv/pose`? Because I still can't find something like `.../pose/position`?

kulkarni-raunak, 2023-11-10:
Look for `/wamv/sensors/position/ground_truth_odometry`

KonradUr, 2023-11-10:
I have no such topic like `/wamv/sensors/position/...`  .  Does anything else need to be done/compiled/whatever after changing the xacro content before relaunching the world?

KonradUr, 2023-11-13:
It started working after installing the latest versions: Gazebo Garden, ROS Humble and vrx. What's more, the current version on my machine shows 90% "simulation speed," while the older version (less than a year old) shows only 60%. So it was worth refreshing the model.  
Thanks for your help.

sontung1010, 2024-03-13:
Hi I added the ground truth flag as above instruction but nothing shows up on my rqt graph for ros2 humble. Can you please help me with this?

kulkarni-raunak, 2024-03-14:
@sontung1010 
Is the topic available for you via `ros2 topic list`?

Maybe post your rqt-graph here.

closed 2025-07-28 by caguero