osrf/vrx #740: gazebo_classic do not have the topic of gps and lidar

FangLian9101, 2023-09-26:
my ros is 20.04 noetic. when I download gazebo_classic .then I run  " roslaunch vrx_gazebo vrx.launch",but I do not have the topic of gps and lidar and other,I only get the topic of camera.
who can help me with this problem.

tejalacc, 2023-10-04:
I am also facing this same issue.

tejalacc, 2023-10-05:
I have installed dependency by using the given command, it solves my issue.
- sudo apt install ros-noetic-hector-gazebo-plugins
- sudo apt install ros-noetic-velodyne-simulator. 
I hope this helps.

YOHyoh131, 2025-02-23:
> I have installed dependency by using the given command, it solves my issue.
> 
> * sudo apt install ros-noetic-hector-gazebo-plugins
> * sudo apt install ros-noetic-velodyne-simulator.
>   I hope this helps.

 I's been with this issue for hours, thank u soooooo much!

caguero, 2025-05-07:
Closing it for now. Please reopen if this is still an issue in the current code.

closed 2025-05-07 by caguero