osrf/vrx #755: Cannot launch gazebo natively after run ros2 launch vrx_gz competition.launch.py world:=sydney_regatta

sontung1010, 2023-12-04:
**Describe the bug**
Cannot launch gazebo natively after run ros2 launch vrx_gz competition.launch.py world:=sydney_regatta. Before it stucks

**Expected behavior**

**To Reproduce**
List the steps to reproduce the problem:
1. Following exactly the tutorial
2. Next, run ''ros2 launch vrx_gz competition.launch.py world:=sydney_regatta"
3. See error
4. Actually, to make gazebo appear with grid as in the screenshot, I have to open another gazebo window before I ran the command above. If not, it will stuck at "requesting a list of world name" after I followed exactly everything in the tutorial.

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 22.04
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Sim 7.0.0
- Graphics Card: Nvidia GTX 3050 Ti
- Are you using VRX or VRX Classic? VRX

**Screenshots**
[
[image]
](url)
[INFO] [launch]: All log files can be found below /home/st/.ros/log/2023-12-04-18-12-18-078183-Vivobook-4592
[INFO] [launch]: Default logging verbosity is set to INFO
['xacro', '/home/st/vrx_ws/install/share/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro', 'namespace:=wamv', 'locked:=true', 'vrx_sensors_enabled:=true', 'thruster_config:=H']
[INFO] [ruby $(which gz) sim-1]: process started with pid [4595]
[INFO] [monitor_sim-2]: process started with pid [4597]
[INFO] [create-3]: process started with pid [4600]
[INFO] [parameter_bridge-4]: process started with pid [4602]
[create-3] [INFO] [1701731538.723301016] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-4] [INFO] [1701731538.731529512] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[parameter_bridge-4] [INFO] [1701731538.732229848] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/task/info (ignition.msgs.Param) -> /vrx/task/info (ros_gz_interfaces/msg/ParamVec)] (Lazy 0)
[parameter_bridge-4] [INFO] [1701731538.733393184] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/debug/wind/speed (ignition.msgs.Float) -> /vrx/debug/wind/speed (std_msgs/msg/Float32)] (Lazy 0)
[parameter_bridge-4] [INFO] [1701731538.733744940] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/debug/wind/direction (ignition.msgs.Float) -> /vrx/debug/wind/direction (std_msgs/msg/Float32)] (Lazy 0)
[create-3] [INFO] [1701731538.924726616] [ros_gz_sim]: Requested creation of entity.
[create-3] [INFO] [1701731538.924772254] [ros_gz_sim]: OK creation of entity.
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/st/.gz/fuel/fuel.gazebosim.org]
[ruby $(which gz) sim-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/st/.gz/fuel/fuel.ignitionrobotics.org]
[INFO] [create-3]: process has finished cleanly [pid 4600]
[INFO] [optical_frame_publisher-5]: process started with pid [4686]
[INFO] [optical_frame_publisher-6]: process started with pid [4688]
[INFO] [optical_frame_publisher-7]: process started with pid [4690]
[INFO] [parameter_bridge-8]: process started with pid [4692]
[INFO] [pose_tf_broadcaster-9]: process started with pid [4694]
[INFO] [robot_state_publisher-10]: process started with pid [4696]
[parameter_bridge-8] [INFO] [1701731539.140072796] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/wamv/pose (ignition.msgs.Pose_V) -> /model/wamv/pose (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.141905518] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/wamv/pose_static (ignition.msgs.Pose_V) -> /model/wamv/pose_static (tf2_msgs/msg/TFMessage)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.142232116] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/joint_state (ignition.msgs.Model) -> /world/sydney_regatta/model/wamv/joint_state (sensor_msgs/msg/JointState)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.143190718] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/contacts (ignition.msgs.Contacts) -> /vrx/contacts (ros_gz_interfaces/msg/Contacts)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.144171674] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/image (ignition.msgs.Image) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/image (sensor_msgs/msg/Image)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.144552926] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/camera_info (ignition.msgs.CameraInfo) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_left_camera_sensor/camera_info (sensor_msgs/msg/CameraInfo)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.145016237] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/image (ignition.msgs.Image) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/image (sensor_msgs/msg/Image)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.145294411] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/camera_info (ignition.msgs.CameraInfo) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/front_right_camera_sensor/camera_info (sensor_msgs/msg/CameraInfo)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.145592254] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan (ignition.msgs.LaserScan) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan (sensor_msgs/msg/LaserScan)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.145958307] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan/points (ignition.msgs.PointCloudPacked) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/lidar_wamv_sensor/scan/points (sensor_msgs/msg/PointCloud2)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.146451545] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/image (ignition.msgs.Image) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/image (sensor_msgs/msg/Image)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.146747624] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/camera_info (ignition.msgs.CameraInfo) -> /world/sydney_regatta/model/wamv/link/wamv/base_link/sensor/middle_right_camera_sensor/camera_info (sensor_msgs/msg/CameraInfo)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.147016872] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/gps_wamv_link/sensor/navsat/navsat (ignition.msgs.NavSat) -> /world/sydney_regatta/model/wamv/link/wamv/gps_wamv_link/sensor/navsat/navsat (sensor_msgs/msg/NavSatFix)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.147362987] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/imu_wamv_link/sensor/imu_wamv_sensor/imu (ignition.msgs.IMU) -> /world/sydney_regatta/model/wamv/link/wamv/imu_wamv_link/sensor/imu_wamv_sensor/imu (sensor_msgs/msg/Imu)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.147687631] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/left/thrust (std_msgs/msg/Float64) -> wamv/thrusters/left/thrust (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.148191599] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/left/pos (std_msgs/msg/Float64) -> wamv/thrusters/left/pos (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.148496475] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/right/thrust (std_msgs/msg/Float64) -> wamv/thrusters/right/thrust (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.148774540] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/right/pos (std_msgs/msg/Float64) -> wamv/thrusters/right/pos (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1701731539.149053796] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [wamv/sensors/acoustics/receiver/range_bearing (ignition.msgs.Param) -> wamv/sensors/acoustics/receiver/range_bearing (ros_gz_interf [... cut ...]
[ruby $(which gz) sim-1] [GUI] [Msg] Loaded plugin [EntityTree] from path [/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/gui/libEntityTree.so]
[ruby $(which gz) sim-1] [GUI] [Dbg] [Application.cc:329] Loading window config
[ruby $(which gz) sim-1] [GUI] [Msg] Using server control service [/server_control]
[ruby $(which gz) sim-1] [GUI] [Dbg] [Application.cc:621] Applying config
[ruby $(which gz) sim-1] [GUI] [Dbg] [MinimalScene.cc:657] Creating gz-rendering interface for OpenGL
[ruby $(which gz) sim-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which gz) sim-1] libEGL warning: egl: failed to create dri2 screen

j-herman, 2023-12-19:
Hi @sontung1010,

Are you still seeing this issue?  If so, I have a few questions that may help with troubleshooting:

Could you post the output you see when you don't open a gazebo window before trying to start VRX?  I'll take a look.
Did you have an earlier version of Gazebo installed on this computer before upgrading to 7.0?  
Are you able to run other worlds that include models from Fuel?

sontung1010, 2023-12-20:
actually, I found a way to work around, thank you.

closed 2023-12-20 by sontung1010

6678lin, 2024-01-13:
Hello, may I ask how you solved this problem? I also met such a situation.

sontung1010, 2024-01-13:
I am not sure, the next time I ran it worked perfectly fine. So maybe do a clean install from the beginning or even reinstall ubuntu