osrf/vrx #761: How to change buoyancy configuration to make my model float

sontung1010, 2023-12-23:
I load my own .stl model into the project and I want to know which parameters to adjust to make my model usable. 

**Describe the bug**
It is sinking too much
[image]

**Expected behavior**
I want it to be visible on the surface like this
[image]

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 22.04
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo Sim 7.0.0
- Are you using VRX or VRX Classic? VRX

caguero, 2024-01-19:
I recommend to go through this tutorial from Gazebo to learn about the `Surface` plugin and how to modify its parameters.

https://gazebosim.org/api/sim/8/surface_vehicles.html

closed 2024-01-19 by caguero

sontung1010, 2024-01-28:
Hello, I got everything worked. But now I am trying to add my function to the repo but I am not sure which is the main root file that everything starts from. Can you please show me that or something like a hierarchy of the program repo? Thank you very much.

Roboticistprogrammer, 2024-07-11:
Hey there @sontung1010, would you mind sharing how you add your .stl model?
I was wandering if I should use **generate_wamv.launch.py** to generate urdf file, or just add my model using /mesh within one of the worlds provided. e.g:
<geometry>
   <mesh filename="package:// ~/Path_To_My_File/XX.stl"/>
</geometry>