osrf/vrx #766: Linear velocity

boeun98, 2024-01-15:
Hello, guys.

I am a vrx beginner.
I want to know the linear velocity of USV.
In what topic can I find out the linear velocity of the USV?

and, using 'rostopic pub' I can publish to thruster the data.
ex) rostopic pub --once /wamv/thrusters/right_thrust_cmd std_msgs/Float32 "data: 1.0"
but I don't know that "data : x.x"
I want to know its unit.

**System Configuration:**
Tell us about your system.
- OS: [Ubuntu 18.04]
- ROS Version: [e.g. ROS, melodic]
- Gazebo Version: [e.g. Gazebo 9.x.x]
- Are you using VRX or VRX Classic? (VRX).

caguero, 2024-01-19:
You need to write a plugin, and then, within your plugin read the world linear velocity component associated to one of its links.

Relevant tutorial:

https://gazebosim.org/api/sim/8/createsystemplugins.html

See an example from VRX where we're reading the linear velocity from a link:

https://github.com/osrf/vrx/blob/main/vrx_gz/src/SimpleHydrodynamics.cc#L207

closed 2024-01-19 by caguero

boeun98, 2024-01-24:
Then, what is the topic '/wamv/sensor/gps/gps/fix_velocity'?

LuoLuoOnCall, 2025-08-14:
Hi, @Polibo1 Did you try to subscribe the /wamv/sensor/gps/gps/fix_velocity to get the velocity? Can that work?