osrf/vrx #768: Fuel world download failed

NordicSailer, 2024-01-25:
**Describe the bug**
sydney-regattta fails to run because fuel world download failed.
Note....I am running for the first time.
**Expected behavior**
Expected the components in the world to be downloaded.

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 22.04
- ROS Version: ROS 2, Humble
- Gazebo Version: 
- Graphics Card: [e.g. Nvidia GTX 650]
-using VRX

ruby $(which ign) gazebo-1] [Wrn] [gz.cc:99] Fuel world download failed because Fetch failed. Other errors
[ruby $(which ign) gazebo-1] Unable to find or download file 
[ERROR] [ruby $(which ign) gazebo-1]: process has died [pid 8517, exit code 255, cmd 'ruby $(which ign) gazebo -v 4 -r  sydney_regatta.sdf --force-version 6'].
[INFO] [launch]: process[ruby $(which ign) gazebo-1] was required: shutting down launched system
[INFO] [parameter_bridge-4]: sending signal 'SIGINT' to process[parameter_bridge-4]
[INFO] [create-3]: sending signal 'SIGINT' to process[create-3]
[INFO] [monitor_sim-2]: sending signal 'SIGINT' to process[monitor_sim-2]
[parameter_bridge-4] [INFO] [1706157169.285433190] [rclcpp]: signal_handler(signum=2)
[create-3] [INFO] [1706157169.285877627] [rclcpp]: signal_handler(signum=2)

caguero, 2024-01-29:
Could you paste the command that you ran?

Also, could you try again right now to discard a temporary problem with Fuel?

I managed to load Sydney Regatta forcing the models to be downloaded.

NordicSailer, 2024-01-30:
The original comment was: 

ros2 launch vrx_gz competition.launch.py world:=sydney_regatta....just copied it from the tutorial.

Thank you for helping with this. So I got it to work. I was running Gazebo Classic with some of the Turtlebot projects. I uninstalled that and the instance of Gazebo Garden. I followed the instructions again and re-loaded Gazebo Garden, and it worked fine.

Apologies to bother with what now is a resolved issue.

closed 2024-01-30 by NordicSailer

Roboticistprogrammer, 2024-07-11:
This is an absolute goldmine, I was working on this for hours and this helped, thank you.