osrf/vrx #773: How can I create more WAM-Vs

Clinz-ZK, 2024-03-06:
May I ask how can I create two or more WAM-Vs on different positions

caguero, 2024-03-11:
You could do it by modifying `vrx_gz/launch/competition.launch.py`:

```diff
+++ b/vrx_gz/launch/competition.launch.py
@@ -43,10 +43,15 @@ def launch(context, *args, **kwargs):
         with open(config_file, 'r') as stream:
             models = Model.FromConfig(stream)
     else:
-      m = Model('wamv', 'wam-v', [-532, 162, 0, 0, 0, 1])
+      m1 = Model('wamv1', 'wam-v', [-532, 162, 0, 0, 0, 1])
       if robot_urdf and robot_urdf != '':
-          m.set_urdf(robot_urdf)
-      models.append(m)
+          m1.set_urdf(robot_urdf)
+      models.append(m1)
+
+      m2 = Model('wamv2', 'wam-v', [-530, 160, 0, 0, 0, 1])
+      if robot_urdf and robot_urdf != '':
+          m2.set_urdf(robot_urdf)
+      models.append(m2)
```

After recompiling and launching, you should see two WAM-Vs now with their own set of topics:

[image]

```bash
/wamv1/joint_states
/wamv1/pose
/wamv1/pose_static
/wamv1/robot_description
/wamv1/sensors/acoustics/receiver/range_bearing
/wamv1/sensors/cameras/front_left_camera_sensor/camera_info
/wamv1/sensors/cameras/front_left_camera_sensor/image_raw
/wamv1/sensors/cameras/front_left_camera_sensor/optical/camera_info
/wamv1/sensors/cameras/front_left_camera_sensor/optical/image_raw
/wamv1/sensors/cameras/front_right_camera_sensor/camera_info
/wamv1/sensors/cameras/front_right_camera_sensor/image_raw
/wamv1/sensors/cameras/front_right_camera_sensor/optical/camera_info
/wamv1/sensors/cameras/front_right_camera_sensor/optical/image_raw
/wamv1/sensors/cameras/middle_right_camera_sensor/camera_info
/wamv1/sensors/cameras/middle_right_camera_sensor/image_raw
/wamv1/sensors/cameras/middle_right_camera_sensor/optical/camera_info
/wamv1/sensors/cameras/middle_right_camera_sensor/optical/image_raw
/wamv1/sensors/gps/gps/fix
/wamv1/sensors/imu/imu/data
/wamv1/sensors/lidars/lidar_wamv_sensor/points
/wamv1/sensors/lidars/lidar_wamv_sensor/scan
/wamv1/shooters/ball_shooter/fire
/wamv1/thrusters/left/pos
/wamv1/thrusters/left/thrust
/wamv1/thrusters/right/pos
/wamv1/thrusters/right/thrust
/wamv2/joint_states
/wamv2/pose
/wamv2/pose_static
/wamv2/robot_description
/wamv2/sensors/acoustics/receiver/range_bearing
/wamv2/sensors/cameras/front_left_camera_sensor/camera_info
/wamv2/sensors/cameras/front_left_camera_sensor/image_raw
/wamv2/sensors/cameras/front_left_camera_sensor/optical/camera_info
/wamv2/sensors/cameras/front_left_camera_sensor/optical/image_raw
/wamv2/sensors/cameras/front_right_camera_sensor/camera_info
/wamv2/sensors/cameras/front_right_camera_sensor/image_raw
/wamv2/sensors/cameras/front_right_camera_sensor/optical/camera_info
/wamv2/sensors/cameras/front_right_camera_sensor/optical/image_raw
/wamv2/sensors/cameras/middle_right_camera_sensor/camera_info
/wamv2/sensors/cameras/middle_right_camera_sensor/image_raw
/wamv2/sensors/cameras/middle_right_camera_sensor/optical/camera_info
/wamv2/sensors/cameras/middle_right_camera_sensor/optical/image_raw
/wamv2/sensors/gps/gps/fix
/wamv2/sensors/imu/imu/data
/wamv2/sensors/lidars/lidar_wamv_sensor/points
/wamv2/sensors/lidars/lidar_wamv_sensor/scan
/wamv2/shooters/ball_shooter/fire
/wamv2/thrusters/left/pos
/wamv2/thrusters/left/thrust
/wamv2/thrusters/right/pos
/wamv2/thrusters/right/thrus
```

closed 2024-03-11 by caguero

Clinz-ZK, 2024-03-18:
very appreciate it for replying my question,it has help me very much.orz

Clinz-ZK, 2024-03-18:
But when I drive the wamvs,how can I  obtain the wamv's real-time position

qingning86, 2025-02-24:
I am currently using ROS Noetic, Gazebo 11.
What I want to ask is how to create more WAM-Vs in ROS 1.
If we follow the above answer and modify the launch file. Can you provide an example.

qingning86, 2025-02-24:
> You could do it by modifying `vrx_gz/launch/competition.launch.py`:
> 
> +++ b/vrx_gz/launch/competition.launch.py
> @@ -43,10 +43,15 @@ def launch(context, *args, **kwargs):
>          with open(config_file, 'r') as stream:
>              models = Model.FromConfig(stream)
>      else:
> -      m = Model('wamv', 'wam-v', [-532, 162, 0, 0, 0, 1])
> +      m1 = Model('wamv1', 'wam-v', [-532, 162, 0, 0, 0, 1])
>        if robot_urdf and robot_urdf != '':
> -          m.set_urdf(robot_urdf)
> -      models.append(m)
> +          m1.set_urdf(robot_urdf)
> +      models.append(m1)
> +
> +      m2 = Model('wamv2', 'wam-v', [-530, 160, 0, 0, 0, 1])
> +      if robot_urdf and robot_urdf != '':
> +          m2.set_urdf(robot_urdf)
> +      models.append(m2)
> After recompiling and launching, you should see two WAM-Vs now with their own set of topics:
> 
> [image]
> 
> /wamv1/joint_states
> /wamv1/pose
> /wamv1/pose_static
> /wamv1/robot_description
> /wamv1/sensors/acoustics/receiver/range_bearing
> /wamv1/sensors/cameras/front_left_camera_sensor/camera_info
> /wamv1/sensors/cameras/front_left_camera_sensor/image_raw
> /wamv1/sensors/cameras/front_left_camera_sensor/optical/camera_info
> /wamv1/sensors/cameras/front_left_camera_sensor/optical/image_raw
> /wamv1/sensors/cameras/front_right_camera_sensor/camera_info
> /wamv1/sensors/cameras/front_right_camera_sensor/image_raw
> /wamv1/sensors/cameras/front_right_camera_sensor/optical/camera_info
> /wamv1/sensors/cameras/front_right_camera_sensor/optical/image_raw
> /wamv1/sensors/cameras/middle_right_camera_sensor/camera_info
> /wamv1/sensors/cameras/middle_right_camera_sensor/image_raw
> /wamv1/sensors/cameras/middle_right_camera_sensor/optical/camera_info
> /wamv1/sensors/cameras/middle_right_camera_sensor/optical/image_raw
> /wamv1/sensors/gps/gps/fix
> /wamv1/sensors/imu/imu/data
> /wamv1/sensors/lidars/lidar_wamv_sensor/points
> /wamv1/sensors/lidars/lidar_wamv_sensor/scan
> /wamv1/shooters/ball_shooter/fire
> /wamv1/thrusters/left/pos
> /wamv1/thrusters/left/thrust
> /wamv1/thrusters/right/pos
> /wamv1/thrusters/right/thrust
> /wamv2/joint_states
> /wamv2/pose
> /wamv2/pose_static
> /wamv2/robot_description
> /wamv2/sensors/acoustics/receiver/range_bearing
> /wamv2/sensors/cameras/front_left_camera_sensor/camera_info
> /wamv2/sensors/cameras/front_left_camera_sensor/image_raw
> /wamv2/sensors/cameras/front_left_camera_sensor/optical/camera_info
> /wamv2/sensors/cameras/front_left_camera_sensor/optical/image_raw
> /wamv2/sensors/cameras/front_right_camera_sensor/camera_info
> /wamv2/sensors/cameras/front_right_camera_sensor/image_raw
> /wamv2/sensors/cameras/front_right_camera_sensor/optical/camera_info
> /wamv2/sensors/cameras/front_right_camera_sensor/optical/image_raw
> /wamv2/sensors/cameras/middle_right_camera_sensor/camera_info
> /wamv2/sensors/cameras/middle_right_camera_sensor/image_raw
> /wamv2/sensors/cameras/middle_right_camera_sensor/optical/camera_info
> /wamv2/sensors/cameras/middle_right_camera_sensor/optical/image_raw
> /wamv2/sensors/gps/gps/fix
> /wamv2/sensors/imu/imu/data
> /wamv2/sensors/lidars/lidar_wamv_sensor/points
> /wamv2/sensors/lidars/lidar_wamv_sensor/scan
> /wamv2/shooters/ball_shooter/fire
> /wamv2/thrusters/left/pos
> /wamv2/thrusters/left/thrust
> /wamv2/thrusters/right/pos
> /wamv2/thrusters/right/thrus

I am currently using ROS Noetic, Gazebo 11.
What I want to ask is how to create more WAM-Vs in ROS 1.
If we follow the above answer and modify the launch file. Can you provide an example.