osrf/vrx #776: Increasing the Maximum Velocity of USV

MustafaBayrak1, 2024-03-27:
**Describe the bug**
Similar to this issue, I am trying to increase the speed of the USV.  
https://github.com/osrf/vrx/issues/731
However, nothing works except changing the parameters in the wamv_gazebo_dynamic_plugin.xacro file. However, it alters the realism of the water surface, as expected. 

I am using VRX Classic.

**Expected behavior**
Trying to increase the maximum velocity the WAM-V can go

**System Configuration:**
Tell us about your system.
- OS: Ubuntu 20.00
- ROS Version: ROS 1
- VRX Classic

MustafaBayrak1, 2024-04-01:
> Did you solve the issue? And I also want to know how to increase the max  speed of wamv on ROS2.
> 

I tried to solve it but it's easier to just migrate to ROS2 at this point. 

To increase the max speed of wamv on ROS2,you need to edit

Clinz-ZK, 2024-04-08:
very thanks for your reply

caguero, 2025-05-07:
Sorry, that original version of ROS 1 and Gazebo classic is no longer supported. Hopefully you were able to use the modern version and this is no longer an issue.

closed 2025-05-07 by caguero