osrf/vrx #783: Spawning multiple robots in the same aquatic environment

ruipaulorocha, 2024-06-12:
Dear VRX developers and maintainers,

As far as I understood, current set of available launch files in the VRX GitHub repository do not support spawning multiple WAM-V robots in Gazebo Garden simulator and pushing the ROS topics and TF tree of each robot to its separate namespace (e.g. robot_0/, robot_1/, etc.). Do you confirm?
Are there any plans to support this in the near future?
Do you have such a multi-robot system example that can be shared in the repository?

jinwoolee1230, 2024-08-19:
have you checked this out?
[https://github.com/osrf/vrx/issues/773](url)

caguero, 2025-05-07:
#773 should describe how to do it. Feel free to reopen the issue if that's not the case.

closed 2025-05-07 by caguero