osrf/vrx #795: How can I use the robot_localization package in ros2?

qingning86, 2025-01-05:
My development environment：
Docker-based Ubuntu 22.04, ROS 2 Humble

My requests:
I want to build raster maps via Cartographer

My question：
There is no odom topic in the VRX project, so I want to simulate the release of odom topic by fusing gps and imu data via the robot_localization package. But I didn't find any tutorials on how to use robot_localization for newbies. What I'm confused about now is how to adjust the configuration file in robot_localization so that it can be adapted to the VRX project and that it can meet my requests as described above. 
And in some of the articles I noticed that the odom topic they posted via robot_localization in ROS 1 were all called “/wamv/robot_localization/odometry/ filtered”. So I'm guessing if the vrx project provides this help in the ROS 1 version but not in the ROS 2 version.

JMANY42, 2025-02-04:
I've been working on something very similar for the past month or so. What I did was use another package to create an odometry frame and then use that in robot_localization. I went for a LiDAR based odometry frame and found [Kiss ICP](https://github.com/PRBonn/kiss-icp) to be very useful and easy to set up, but any other solution should work. After that, I included the correct odometry frame in the robot_localization param file and changed the topic names of sensors messages to the ones produced by the vrx package. 

If there's anything else I can help with let me know!

qingning86, 2025-02-16:
> I've been working on something very similar for the past month or so. What I did was use another package to create an odometry frame and then use that in robot_localization. I went for a LiDAR based [...]

Thank you for your reply! It was very helpful to me! I successfully published the /kiss/odometry topic using the Kiss ICP project, and will be merging this data in robot_localization next. However, I still don't know much about how to use the robot_localization package at this point, and was wondering if you could provide some tutorials accordingly.
Also, I noticed that in ros2, the lidar detects the sea surface, which is shown as a line in rviz. I'm wondering if this bug will have an effect on the raster maps built if you have successfully built them in a ros2 based vrx project.
Again, thanks for your previous help and look forward to your reply!

qingning86, 2025-02-17:
> I've been working on something very similar for the past month or so. What I did was use another package to create an odometry frame and then use that in robot_localization. I went for a LiDAR based [...]

Perhaps I misunderstood you. I tweaked some of the parameters of the launch file in kiss-ICP and have now successfully released the tf coordinate transformation from odom_lidar to wamv/wamv/base_link. I'm thinking that maybe I don't need to use the robot_localization package anymore and just use the odometry information provided by kiss-ICP to build the map. However, when I use the Cartographer tool for map building, the map coordinate system never posts properly.