osrf/vrx #798: Not failed run sydney_regatta

XYjy949, 2025-02-19:
Hi,guys！

When I run the following command and this error occurs, how can I solve it? Have you ever encountered this problem before？

environment:ros2 humble ,gazebo garden 

$ros2 launch vrx_gz competition.launch.py world:=sydney_regatta

[INFO] [launch]: All log files can be found below /home/cczh/.ros/log/2025-02-19-16-57-25-870788-cczhros2-9819
[INFO] [launch]: Default logging verbosity is set to INFO
['xacro', '/home/cczh/vrx_ws/install/share/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro', 'namespace:=wamv', 'locked:=true', 'vrx_sensors_enabled:=true', 'thruster_config:=H']
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Error Code 30: Msg: Error parsing XML from string: Error=XML_ERROR_PARSING_TEXT ErrorID=8 (0x8) Line number=26
Error Code 30: Msg: Unable to read SDF string: The 'gz' command provides a command line interface to the Gazebo Tools.

wojtess, 2025-04-13:
What OS do you use?

caguero, 2025-05-02:
@XYjy949 , could you clarify your OS, Gazebo version, ROS version, installation method (locally or using Docker)?

bsb808, 2025-05-07:
@caguero I run into what appears to be the same issue.

My environment is as follows:
* Host OS: Ubuntu 24.04
* Install and run using dockwater project, specifically the humble Dockerfile from the main branch of the dockwater repository. Following this tutorial: https://github.com/osrf/vrx/wiki/docker_install_tutorial 
* The build within docker executes without errors.  Followed https://github.com/osrf/vrx/wiki/installation_tutorial
* Finally ran the demo with the command from here: https://github.com/osrf/vrx/wiki/running_vrx_tutorial

Here is the first part of the output from the command line:
`
bsb@8b673f9e655d:~/vrx_ws$ ros2 launch vrx_gz competition.launch.py world:=sydney_regatta
[INFO] [launch]: All log files can be found below /home/bsb/.ros/log/2025-05-06-18-40-42-199241-8b673f9e655d-1687
[INFO] [launch]: Default logging verbosity is set to INFO
['xacro', '/home/bsb/vrx_ws/install/share/wamv_gazebo/urdf/wamv_gazebo.urdf.xacro', 'namespace:=wamv', 'locked:=true', 'vrx_sensors_enabled:=true', 'thruster_config:=H']
[INFO] [ruby $(which gz) sim-1]: process started with pid [1690]
[INFO] [monitor_sim-2]: process started with pid [1692]
[INFO] [create-3]: process started with pid [1694]
[INFO] [parameter_bridge-4]: process started with pid [1697]
[create-3] [INFO] [1746582042.909269728] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-4] [INFO] [1746582042.909358049] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[parameter_bridge-4] [INFO] [1746582042.911150973] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/task/info (ignition.msgs.Param) -> /vrx/task/info (ros_gz_interfaces/msg/ParamVec)] (Lazy 0)
[parameter_bridge-4] [INFO] [1746582042.916473203] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/debug/wind/speed (ignition.msgs.Float) -> /vrx/debug/wind/speed (std_msgs/msg/Float32)] (Lazy 0)
[parameter_bridge-4] [INFO] [1746582042.917694992] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/vrx/debug/wind/direction (ignition.msgs.Float) -> /vrx/debug/wind/direction (std_msgs/msg/Float32)] (Lazy 0)
[ruby $(which gz) sim-1] The 'gz' command provides a command line interface to the Gazebo Tools.
[ruby $(which gz) sim-1] 
[ruby $(which gz) sim-1]   gz <command> [options]
[ruby $(which gz) sim-1] 
[ruby $(which gz) sim-1] List of available commands:

...

[ruby $(which gz) sim-1]   --commands                 Show the available commands.
[ruby $(which gz) sim-1] Use 'gz help <command>' to print help for a command.
[ERROR] [ruby $(which gz) sim-1]: process has died [pid 1690, exit code 255, cmd 'ruby $(which gz) sim -v 4 -r  sydney_regatta.sdf --force-version 7'].
[create-3] [INFO] [1746582047.909879687] [ros_gz_sim]: Requesting list of world names.
[create-3] [INFO] [1746582052.910078465] [ros_gz_sim]: Requesting list of world names.

`

The full output is attached.

[vrx.log](https://github.com/user-attachments/files/20075303/vrx.log)

pedropmp, 2025-05-07:
I am facing the same issue, but Host OS is Ubuntu 22.04. Used the same dockwater project.

pedropmp, 2025-05-07:
@caguero It worked!

[image]

```[parameter_bridge-8] [INFO] [1746629058.930980876] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/sydney_regatta/model/wamv/link/wamv/imu_wamv_link/sensor/imu_wamv_sensor/imu (ignition.msgs.IMU) -> /world/sydney_regatta/model/wamv/link/wamv/imu_wamv_link/sensor/imu_wamv_sensor/imu (sensor_msgs/msg/Imu)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.931344670] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/left/thrust (std_msgs/msg/Float64) -> wamv/thrusters/left/thrust (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.931808252] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/left/pos (std_msgs/msg/Float64) -> wamv/thrusters/left/pos (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.932032698] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/right/thrust (std_msgs/msg/Float64) -> wamv/thrusters/right/thrust (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.932312144] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [wamv/thrusters/right/pos (std_msgs/msg/Float64) -> wamv/thrusters/right/pos (ignition.msgs.Double)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.932540790] [wamv.ros_gz_bridge]: Creating GZ->ROS Bridge: [wamv/sensors/acoustics/receiver/range_bearing (ignition.msgs.Param) -> wamv/sensors/acoustics/receiver/range_bearing (ros_gz_interfaces/msg/ParamVec)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.932827765] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [/pinger/set_pinger_position (geometry_msgs/msg/Vector3) -> /pinger/set_pinger_position (ignition.msgs.Vector3d)] (Lazy 0)
[parameter_bridge-8] [INFO] [1746629058.933171949] [wamv.ros_gz_bridge]: Creating ROS->GZ Bridge: [/wamv/shooters/ball_shooter/fire (std_msgs/msg/Bool) -> /wamv/shooters/ball_shooter/fire (ignition.msgs.Boolean)] (Lazy 0)
[robot_state_publisher-10] [WARN] [1746629058.934607085] [kdl_parser]: The root link wamv/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-10] [INFO] [1746629058.934703164] [wamv.robot_state_publisher]: got segment wamv/ball_shooter_base_link
[robot_state_publisher-10] [INFO] [1746629058.934743383] [wamv.robot_state_publisher]: got segment wamv/ball_shooter_launcher_link
[robot_state_publisher-10] [INFO] [1746629058.934747533] [wamv.robot_state_publisher]: got segment wamv/base_link
[robot_state_publisher-10] [INFO] [1746629058.934750973] [wamv.robot_state_publisher]: got segment wamv/cpu_cases_link
[robot_state_publisher-10] [INFO] [1746629058.934753963] [wamv.robot_state_publisher]: got segment wamv/front_left_camera_link
[robot_state_publisher-10] [INFO] [1746629058.934756943] [wamv.robot_state_publisher]: got segment wamv/front_left_camera_link_optical
[robot_state_publisher-10] [INFO] [1746629058.934759793] [wamv.robot_state_publisher]: got segment wamv/front_left_camera_post_arm_link
[robot_state_publisher-10] [INFO] [1746629058.934762663] [wamv.robot_state_publisher]: got segment wamv/front_left_camera_post_link
[robot_state_publisher-10] [INFO] [1746629058.934765393] [wamv.robot_state_publisher]: got segment wamv/front_right_camera_link
[robot_state_publisher-10] [INFO] [1746629058.934768123] [wamv.robot_state_publisher]: got segment wamv/front_right_camera_link_optical
[robot_state_publisher-10] [INFO] [1746629058.934770933] [wamv.robot_state_publisher]: got segment wamv/front_right_camera_post_arm_link
[robot_state_publisher-10] [INFO] [1746629058.934773602] [wamv.robot_state_publisher]: got segment wamv/front_right_camera_post_link
[robot_state_publisher-10] [INFO] [1746629058.934776332] [wamv.robot_state_publisher]: got segment wamv/gps_wamv_link
[robot_state_publisher-10] [INFO] [1746629058.934779072] [wamv.robot_state_publisher]: got segment wamv/imu_wamv_link
[robot_state_publisher-10] [INFO] [1746629058.934781732] [wamv.robot_state_publisher]: got segment wamv/left_battery_link
[robot_state_publisher-10] [INFO] [1746629058.934784592] [wamv.robot_state_publisher]: got segment wamv/left_engine_link
[robot_state_publisher-10] [INFO] [1746629058.934787402] [wamv.robot_state_publisher]: got segment wamv/left_propeller_link
[robot_state_publisher-10] [INFO] [1746629058.934789952] [wamv.robot_state_publisher]: got segment wamv/lidar_wamv_link
[robot_state_publisher-10] [INFO] [1746629058.934792752] [wamv.robot_state_publisher]: got segment wamv/lidar_wamv_post_arm_link
[robot_state_publisher-10] [INFO] [1746629058.934795472] [wamv.robot_state_publisher]: got segment wamv/lidar_wamv_post_link
[robot_state_publisher-10] [INFO] [1746629058.934798102] [wamv.robot_state_publisher]: got segment wamv/middle_right_camera_link
[robot_state_publisher-10] [INFO] [1746629058.934800642] [wamv.robot_state_publisher]: got segment wamv/middle_right_camera_link_optical
[robot_state_publisher-10] [INFO] [1746629058.934803272] [wamv.robot_state_publisher]: got segment wamv/middle_right_camera_post_arm_link
[robot_state_publisher-10] [INFO] [1746629058.934806102] [wamv.robot_state_publisher]: got segment wamv/middle_right_camera_post_link
[robot_state_publisher-10] [INFO] [1746629058.934808702] [wamv.robot_state_publisher]: got segment wamv/receiver
[robot_state_publisher-10] [INFO] [1746629058.934811382] [wamv.robot_state_publisher]: got segment wamv/right_battery_link
[... cut ...]
[ruby $(which gz) sim-1] [Wrn] [Ogre2Camera.cc:412] Ogre2Camera::SetVisibilityMask: Mask bits c0000000 are set but will be ignored as they conflict with the reserved bits used internally by the ogre2 backend.
[ruby $(which gz) sim-1] [Wrn] [Ogre2Camera.cc:412] Ogre2Camera::SetVisibilityMask: Mask bits c0000000 are set but will be ignored as they conflict with the reserved bits used internally by the ogre2 backend.
[ruby $(which gz) sim-1] [Wrn] [Ogre2Camera.cc:412] Ogre2Camera::SetVisibilityMask: Mask bits c0000000 are set but will be ignored as they conflict with the reserved bits used internally by the ogre2 backend.
```

caguero, 2025-05-07:
Great! In any case I just merged #807 and updated the wiki. It should work with the modern environment already. Closing the issue but reopen if needed.

closed 2025-05-07 by caguero