osrf/vrx #799: The URL in VRX Classic tutorial cannot be opened

qingning86, 2025-02-21:
At the following website“ https://github.com/osrf/vrx/wiki/vrx_classic_customizing_vrx_tutorials ”. The redirect link 'Visualizing with RViz:' located at its bottom is not functioning properly. The URL of the redirect link is“ https://github.com/osrf/vrx/wiki/tutorials-Visualizing%20with%20RViz ”.When I click on it, it will jump to“ https://github.com/osrf/vrx/wiki/ ”. And display 'You do not have permission to update this wiki.' at the top of the page.

The content in this tutorial is very important to me as I need to use the robot_localization package. I hope this link can be fixed to normal use as soon as possible.

nicholasduffy, 2025-02-28:
Is [this](https://github.com/osrf/vrx/wiki/vrx_classic_rviz_tutorial) the link you were looking for? It seems to be linked incorrectly as you mentioned, however I just used the Pages dropdown search to find it.

[image]

qingning86, 2025-03-01:
> Is [this](https://github.com/osrf/vrx/wiki/vrx_classic_rviz_tutorial) the link you were looking for? It seems to be linked incorrectly as you mentioned, however I just used the Pages dropdown search to find it.
> 
> [image]

Thank you for your reply, but the page I am looking for is not the one you mentioned. I used a page search tool to find its previous version and attached the URL“ https://web.archive.org/web/20200916134658/https://github.com/osrf/vrx/wiki/tutorials -Visualizing%20with%20RViz”

closed 2025-03-01 by qingning86