osrf/vrx #800: Error reported when using command 'colcon build --merge-install‘

wf789456123, 2025-03-12:
**Describe the bug**
In the (https://github.com/osrf/vrx/wiki/installation_tutorial) tutorial, using this command is feasible, but there is a missing error message during runtime: 
CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

I think this is very strange because I am using ROS2 Humble and Gazebo Garden.
I tried to install Catkin, but encountered an error:
sudo apt-get install ros-humble-catkin
[sudo] password for wf: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-catkin

**Expected behavior**
Successfully install VRX.

**To Reproduce**
List the steps to reproduce the problem:
1. Run '...'
2. Next, run '....'
3. See error

**System Configuration:**

- OS: Ubuntu 22.04 LTS
- ROS Version: ROS 2, Humble
- Gazebo Version: Gazebo garden
- Graphics Card: Nvidia GTX 3090
- Are you using VRX or VRX Classic? (The default is VRX).

**Screenshots**
:~/vrx_ws$ colcon build --merge-install
Starting &gt;&gt;&gt; wave_gazebo_plugins
Starting &gt;&gt;&gt; usv_msgs
Starting &gt;&gt;&gt; wamv_description
--- stderr: wave_gazebo_plugins
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake &lt; 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument &lt;min&gt; value or use a ...&lt;max&gt; suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at CMakeLists.txt:25 (cmake_policy):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   &lt;&lt;&lt; wave_gazebo_plugins [0.06s, exited with code 1]
Aborted  &lt;&lt;&lt; usv_msgs [0.06s]
Aborted  &lt;&lt;&lt; wamv_description [0.06s]

Summary: 0 packages finished [0.21s]
  1 package failed: wave_gazebo_plugins
  2 packages aborted: usv_msgs wamv_description
  3 packages had stderr output: usv_msgs wamv_description wave_gazebo_plugins
  7 packages not processed

**Additional notes**
Add any other context about the problem here.

wojtess, 2025-04-13:
Did you executed `source /opt/ros/humble/setup.bash` before trying to build?

caguero, 2025-05-02:
As @wojtess  mentioned, you probably forgot to source your Humble `setup.bash`. Please, reopen if you still experience an issue.

closed 2025-05-02 by caguero