osrf/vrx #802: Launch problem ROS2 Jazzy

MF-Ahmed, 2025-03-24:
Hi, 
I have build the dev-jazzy branch. And getting the following error while launch 
`ros2 launch vrx_gz vrx_environment.launch.py world:=nbpark     
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-03-24-15-32-36-054063-pc-ls2n-3571-59510
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - ModuleNotFoundError: No module named 'sdformat13'
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown
`
Thanks  a lot

HDenial, 2025-04-28:
were you able to fix it? i also needed to run it on  Jazzy+gz-Harmonic

caguero, 2025-05-02:
This issue occurs because you're installing Gazebo directly from the ROS deb package repository and unfortunately, the Python bindings are disabled there. In other words, you're leveraging ROS 2 Gazebo vendor packages. See the [current issue about the Python bindings](https://gazebosim.org/docs/latest/ros2_gz_vendor_pkgs/).

The alternative is to install Gazebo directly from `packages.osrfoundation.org` following [the Gazebosim tutorial](https://gazebosim.org/docs/harmonic/install_ubuntu/#binary-installation-on-ubuntu). If you're using ROS 2 Jazzy, you probably want to pair it with Gazebo Harmonic.

As a side note, we're updating VRX to support Ubuntu 24.04, ROS 2 Jazzy and Gazebo Harmonic. See[ this pull request](https://github.com/osrf/vrx/pull/807).