osrf/vrx #839: jazzy branch possibly broken

nzlz, 2025-07-07:
Hi!

First of all, slightly off-topic, it would be nice to have at least 1 test launching the system. Right now only compilation is tested and issues with vendor and not vendor packages are not. (from what i can understand)

My current issue trying to rebase on vrx/jazzy is that [](http://packages.osrfoundation.org/gazebo/ubuntu-stable) provides sdformat `14.8.0` and the installed vendor package comes with `14.7.0`.

I see [https://github.com/gazebo-release/sdformat_vendor/pull/15](https://github.com/gazebo-release/sdformat_vendor/pull/15) , which is not merged. This means that when trying to launch the sw stack i get incompatible symbols.

Summary:
- gz-harmonic shipped with (http://packages.osrfoundation.org/gazebo/ubuntu-stable) depends on sdformat, and installs python bindings for `14.8.0` .
- ros-jazzy-desktop-full now installs vendor at `14.7.0` via:
    ```
    ros-jazzy-desktop-full 
      └── ros-jazzy-ros-gz-sim-demos 
          └── ros-jazzy-ros-gz-sim 
              └── ros-jazzy-sdformat-vendor
    ```

There are [many deps on vendor on the ros-gz codebase](https://github.com/gazebosim/ros_gz/blob/5995c031fe06b75ad3a1304c7a88323609b5b119/ros_gz_sim/package.xml#L30-L33)... so I don't thinkg building from source on a custom branch is a valid solution here.

Thoughts? @caguero @j-rivero

caguero, 2025-07-08:
Thanks @nzlz . I don't think you're supposed to install `ros-jazzy-sdformat-vendor`. Mainly because it doesn't have Python bindings enabled. You can see in the `Dockerfile` how we install `gz-harmonic` from the `osrfoundation.org` package repository.

persnacon, 2025-07-11:
I'm experiencing a similar issue. It was working fine until last month, but after a fresh installation, it's no longer launching.

Here's the sequence of events:

1. Completed installation of ROS2 Jazzy + Gazebo Harmonic.
2. Installed osrf/vrx.
3. Upon launch, the following messages are displayed:

```
    [INFO] [launch]: All log files can be found below /home/username/.ros/log/2025-07-11-13-42-19-684968-hj-960XFH-89013
    [INFO] [launch]: Default logging verbosity is set to INFO
    [ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'ros_gz_sim' not found, searching: ['/home/username/vrx_ws/install', '/opt/ros/jazzy']"
```

 4. After installing ros-jazzy-ros-gz-sim using the `sudo apt-get install ros-jazzy-ros-gz-sim` command, I tried to launch again, and the following messages appeared:

```
    [INFO] [launch]: All log files can be found below /home/username/.ros/log/2025-07-11-13-43-57-237217-hj-960XFH-90145
    [INFO] [launch]: Default logging verbosity is set to INFO
    [ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
    - ImportError: /usr/lib/python3/dist-packages/sdformat14.cpython-312-x86_64-linux-gnu.so: undefined symbol: _ZNK3sdf3v144Cone14AxisAlignedBoxEv
    - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown
```

Please help. Is there a solution to this?

persnacon, 2025-07-11:
> I'm experiencing a similar issue. It was working fine until last month, but after a fresh installation, it's no longer launching.
>
> Here's the sequence of events:
>
>     1. Completed installation of [...]

Great news! I'm glad you found a solution after all that effort.

Here's the English translation of your message:

After a lot of struggle, I found a solution!

```
mkdir -p ~/vrx_ws/src
cd ~/vrx_ws/

vcs import --input https://raw.githubusercontent.com/gazebo-tooling/gazebodistro/master/collection-harmonic.yaml .

git clone https://github.com/osrf/vrx.git -b jazzy
git clone https://github.com/gazebosim/ros_gz.git -b jazzy

cd ~/vrx_ws/
export GZ_BUILD_FROM_SOURCE=1
rosdep install --from-paths src --ignore-src -r -y
source /opt/ros/jazzy/setup.bash
colcon build --symlink-install
```

With this method, everything runs without any issues, including launching.

j-rivero, 2025-07-11:
> First of all, slightly off-topic, it would be nice to have at least 1 test launching the system. Right now only compilation is tested and issues with vendor and not vendor packages are not. (from what i can understand)

Fully agree. We did not have the resources or time for this before but would be great if we can implement it soon. Feel free to open a PR if you can contribute.

> I see [gazebo-release/sdformat_vendor#15](https://github.com/gazebo-release/sdformat_vendor/pull/15) , which is not merged. This means that when trying to launch the sw stack i get incompatible symbols.

I've bloomed the PR. It will take a bit to land in the stable repositories. The binaries should not be incompatible within a major version release but I think that in this case we are speaking about the Python bindings <-> sdformat libs which it is a different thing that mixing headers + sources in C++ in the same major version.

> looks like the sdformat-vendor 14.8.0 package was just released: [gazebo-release/sdformat_vendor#15](https://github.com/gazebo-release/sdformat_vendor/pull/15), thanks [@j-rivero](https://github.com/j-rivero)!

Yep, it looks like it is now in the testing repository http://packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-jazzy-sdformat-vendor/ . If someone can add the [testing repository](https://docs.ros.org/en/rolling/Installation/Testing.html) to the image and check that things are fixed, would be great. 

> After a lot of struggle, I found a solution!

That is a good way. In fact I was working yesterday on getting rid of the vendor packages and make the full image to work with .deb packages for Gazebo. Did not finish to try it yet but my work is in the [`jrivero/no_vendor_gz`](https://github.com/osrf/vrx/compare/jrivero/no_vendor_gz?expand=1) branch. Will try to back to it early next week.

j-rivero, 2025-07-17:
I think that I finally found a solution that can scale to other Gz libraries with Python packages, can work without requiring the vendor repositories to be attended and transparently provide the user with the same experience.

If you guys can give a try to #840, would be great.