osrf/vrx #846: Plugin Problem Ros Humble Gazebo Garden

nadia-97, 2025-07-24:
[Err] [SystemLoader.cc:125] Failed to load system plugin [gz::sim::systems::ForceTorque] : could not instantiate from library [gz-sim-forcetorque-system] from path [/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-forcetorque-system.so].

I installed vrx using the humble branch as I am using ros2 humble with gazebo garden and ubuntu 22.04. I am facing an issue with some plug ins, one of them is mentioned above. The simulation launches with the command mentioned below and in the simulation the water is not there and the sky glitches, I am assuming these are due to the important plug ins not being found. 

command: ros2 launch vrx_gz competition.launch.py world:=sydney_regatta

nadia-97, 2025-07-25:
> Please, take a look at the updated instructions. VRX now [requires](https://github.com/osrf/vrx/wiki/system_requirements) ROS 2 Jazzy, Gazebo Harmonic, and Ubuntu 24.04. Alternatively, you could use the `humble` branch that should work with Gazebo Garden.

I did use the humble branch. And upon your suggestion I now tried with ubuntu 24 with ros2 jazzy and gazebo harmonic but I am getting the following error upon launching with the same command. 

ros2 launch vrx_gz competition.launch.py world:=sydney_regatta

[INFO] [launch]: All log files can be found below /home/nadia/.ros/log/2025-07-25-11-58-08-985843-nadia-pc-30036
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - ImportError: /usr/lib/python3/dist-packages/sdformat14.cpython-312-x86_64-linux-gnu.so: undefined symbol: _ZNK3sdf3v144Cone14AxisAlignedBoxEv
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

caguero, 2025-07-25:
We just released 3.0.4 of VRX with the patch to fix the issue. Are you using that version?