osrf/vrx #849: I have error with "sdformat14"

dh392, 2025-07-30:
I'm using ubuntu 24.04, ros 2 jazzy and gazebo harmonic. 

But I have error when run command "ros2 launch vrx_gz competition.launch.py world:=sydney_regatta" :

[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - ImportError: /usr/lib/python3/dist-packages/sdformat14.cpython-312-x86_64-linux-gnu.so: undefined symbol: _ZNK3sdf3v144Cone14AxisAlignedBoxEv
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

How to fix this error? Thanks

caguero, 2025-08-05:
Hi @dh392 , are you using Docker or running locally?

j-rivero, 2025-09-01:
This problem is the same discussed and fixed for docker in https://github.com/osrf/vrx/issues/839.

The solution for this particular case of sdformat14 was released (synced) into the main ROS 2 Jazzy repository in the [sync of Aug 20th](https://discourse.openrobotics.org/t/new-packages-and-patch-release-for-jazzy-jalisco-2025-08-20/49667). The package [ros-jazzy-sdformat-vendor](https://github.com/gazebosim/sdformat): was updated from 0.0.9-1 → 0.0.10-1. That should bring the `14.8.0` version into the ROS packages, the same than in the system.

I just install the software on my local machine and the problem is not here anymore.

> out of curiosity what is the more "perma" fix if there is one rather than having something in your bash rc? Thank you for the fix, appreciate it.

Yes, there is a more permanent solution detailed in https://github.com/gazebo-tooling/gz_vendor/issues/15 that is to solve https://github.com/gazebo-tooling/gz_vendor/issues/2.