osrf/vrx #853: dynamic parameters are different from the published paper

LuoLuoOnCall, 2025-08-12:
Hi, I want to design a MPC controller, so I need the dynamic parameters.

And I found the parameters in the code seem different from that which you described in your paper.

in the paper:

[image]

In SimpleHydrodynamics.cc

the Drag is:
```
Dmat(0, 0) = this->dataPtr->paramXu +
    this->dataPtr->paramXuu * std::abs(localLinearVel.X());
  Dmat(1, 1) = this->dataPtr->paramYv +
    this->dataPtr->paramYvv * std::abs(localLinearVel.Y());
  Dmat(2, 2) = this->dataPtr->paramZw +
    this->dataPtr->paramZww * std::abs(localLinearVel.Z());
  Dmat(3, 3) = this->dataPtr->paramKp +
    this->dataPtr->paramKpp * std::abs(localAngularVel.X());
  Dmat(4, 4) = this->dataPtr->paramMq +
    this->dataPtr->paramMqq * std::abs(localAngularVel.Y());
  Dmat(5, 5) = this->dataPtr->paramNr +
    this->dataPtr->paramNrr * std::abs(localAngularVel.Z());
```
The parameters like Nvr, which it is supposed to be used in Dmat(5,5), are missing.

But I noticed in the paper you say:

This simplified six degree-of-freedom model and the Gazebo plugin are generally applicable to surface vessels. Applying this model to a specific vessel requires determining values for the hydrodynamic derivatives. These values can be approximated from first principles (e.g., potential flow theory) or estimated through experimental testing (e.g., scale model testing). In [[25]] the authors estimate the hydrodynamic derivatives for a three degree-of-freedom model of the WAM-V vessel by manually tuning the coefficients so that the model outputs agree with experimental measurements from at-sea maneuvering tests. Each of the of linear drag terms in [(21)] are estimated, **but only the surge term in the quadratic drag matrix [(22)] is identified, implying that the remaining quadratic terms are neglected**. These specific values are adopted for modeling the WAM-V within the simulation as part of the VRX challenge use-case.

I don't know much about dynamic. And I want to know if  there is anything wrong or it is neglected as 0, so it don't need to be considered in the code.

bsb808, 2025-08-27:
Hi @LuoLuoOnCall - thank you for the question.

The values for the coefficients have been modified over the years to try and improve the fidelity of the vessel motion compared to experiments.   The stock vessel in VRX is an exemplar - a digital cousin - of a WAM-V with a generic propulsion system.

If you are going to be using MPC on hardware it will be important that your team does the system identification to estimate the various rigid-body, hydrostatic, hydrodynamic and propulsion parameters.  Also, for operating in coastal/harbor environments the wind response (windage) is a critical factor.   Each WAM-V build is different (there is no stock options for propulsion, buoyancy, payload, etc.) - so using MPC you'll want to have coefficients that are experimentally derived for your particular configuration.

I'm closing the issue, but feel free to re-open if necessary.

LuoLuoOnCall, 2025-08-28:
Hi,@bsb808. thank you so much for your reply. Actually, I just want to test my MPC controller in VRX. As we all know, MPC can hardly  be truly applied on USV in the real world.

And I also noticed  that in https://github.com/osrf/vrx/blob/jazzy/vrx_urdf/wamv_gazebo/urdf/dynamics/wamv_gazebo_dynamics_plugin.xacro
you set the added mass as 0.

Does that also improve the fidelity of the vessel motion compared to experiments?

I'm a little confused about that because it looks strange.