osrf/vrx #858: Problem launching simulator in Humble or Harmonic branch

saspragkathos, 2025-09-28:
I am trying to build and run the vrx simulator in Ubuntu 22.04 setup with ROS2 Humble. because that's the development setup of my frameworks right now.
I have cloned both humble and harmonic branches they being built properly but when I run ros2 launch to launch the sydney_regatta which is in the tutorial a problem occurs regarding 1.11 to 1.10 sdf conversion while not file seems to have this version.
Is anybody that has the same problem?
I have tested it both in native Ubuntu 22.04 and in WSL2 interpretation.

saspragkathos, 2025-09-28:
Also why in the harmonic branch the vrx_gz has a dependency on gz-sim7 which is the Gazebo Garden?
And as I am checking all the compatibilities with the rest gz libraries follows gazebo garden as well?
Should it be gz-sim8, sdformat14 etc etc?

caguero, 2025-09-28:
The `humble` branch should be paired with Gazebo Garden.

closed 2025-10-30 by caguero